openbikesensor/OpenBikeSensor3dPrintableCase #14: More clearance for magnet holes

gluap, 2021-05-13:
It would be great to add .1mm to .2mm of clearance to the magnet holes in all directions. Currently the holes are designed exactly to dimension of the magnets. But in 3D printing, especially at higher layer heights used by most people for larger objects this makes it very hard to slide the magnets in. The reason is that even with a well-calibrated printer the edges of each layer resemble a semicircle (or circular sector). And while the "average edge of deposited material" will be on-spot for a perfectly calibrated printer, the bellies of the circles or circle-sectors unavoidably protude beyond the planned edge of the layer, making the hole slightly smaller. 

[image]

I know that this can be compensated for in some slicers ("enlarge holes in layers" type settings). However the setting it does not apply to all orientations of printing (because in some orientations the slicer does not detect the magnet holes as "holes" as they are not holes in the topographic sense. E.g. the pocket for the magnet in the display case.).

Mario (who is printing most of our cases) has locally modified the models to have more clearance, but it would be great if there was a bit of space to wiggle the magnets in the upstream model, as it would avoid a lot of test-prints on the user side. While a too-big hole can be compensated for by a bit of cyanacrylate or even a leftover piece of brim wedged in on the side, a too-small hole is hard to enlarge when it is only 3mm wide and its opening even smaller.

Alternatively if there is a proven method to wiggle the magnets in, this could be documeneted. The Issue I have is that Neodymium magnets aren't supposed to be heated above the 80°C that are needed to soften PETG.

I would have put this in terms of a pull request but I believe the current version of the Fusion360 files aren't in the repo yet.

opatut, 2021-05-13:
I totally agree.

> Neodymium magnets aren't supposed to be heated above the 80°C that are needed to soften PETG

I've heard of people successfully "soldering" them in place though. I can't promise it, but apparently for a short duration it doesn't totally wreck them.

> I would have put this in terms of a pull request but I believe the current version of the Fusion360 files aren't in the repo yet.

It's difficult, yeah. The latest modifications were, I think, done in Inventor from importing the step model, or something like that. There is sadly zero support for a PR-based workflow with these software packages, even exporting and importing fusion projects is a pain. We're looking into alternatives. Do you know of any good ones?

gluap, 2021-05-13:
@opatut No, I don't know a better solution for versioning 3d models. Unless one manages them in the workflow of one of the software packages (I believe fusion allows collaboration on models, but have no experience in using it), versioning of 3D models mostly comes down to managing binary blobs. Merging is impossible so the workflow with the least overhead is when one person manages the model. Some VCSes allow exclusive checkouts for this kind of work to ensure only one person at a time can work on un-mergable file format. Git is not one of them though.

Another possible path would be to build the model entirely in openscad (where everything is source code and thus merges are easy). But in my experience that becomes painful because of very long render times when one approaches the complexity of the cases. Also the learning curve of openscad is rather steep.

tobst, 2021-05-20:
Attempted to fix by  73d93de4da1ab3c2d892554190c8c89588ca5360 
However I see the risk of tuning it just to my printer now.

tobst, 2021-05-20:
I added 0.1mm in all directions, but magnets can be pulled out with the counterpart now.

gluap, 2021-05-20:
With 0.15mm I added to the step file for experiments I observed that magnets could be pulled out, too. As stated above however I think a hole that works on most printers but needs glue (or a bit of brim wedged in) is preferrable to one that needs lots of test prints going to the garbage on most printers. Ill print the update tonight to see how it works out here. 0.1mm sounds snug but still insertable. Before I wasn't able to build the display cases at all without modification, the magnet just wouldn't drop into the recess no matter what.

tobst, 2021-05-20:
Should we supply an scad script to extract the critical part into a small test object which can be printed in a few minutes?

gluap, 2021-05-20:
I usually cut the interesting sections out in the slicer. If the model position can be assumed as fixed though, an openscad script could be provided that directly enlarges magnet holes parametrically. It could even be used to generate a set of test pieces with different clearances.

gluap, 2021-05-20:
An alternative solution that just sprang into my mind would be to enlarge the clearance but leave a very narrow notch to wedge the magnet against, which could hold it but would require less force to insert it. Crude ascii-art for illustration of a square magnet hole below:

```
+---v---+
|       |
>       <
|       |
+---^---+

gluap, 2021-05-23:
Following up on @opatut's lead about the magnets surviving "soldering" into the plastic: according to [supermagnete.de](https://www.supermagnete.de/faq/Wie-heiss-duerfen-Magnete-werden), the cubic magnets can be heated above 80° without damaging them thanks to their favourable aspect ratio. For the flat magnets in the display cases heat may be more of an issue and partial loss of magnetization is expected.

tobst, 2021-05-24:
After testing diferent clearances and your suggeste shape I am in favor for a script with default coordinates but different ones as an option to modify it to your needs.

tobst, 2021-05-24:
Event heating them up is a difficult task becaus they will always stick to each other or your heated metal tool.

gluap, 2021-06-19:
Regarding the cubic magnets of the main case:

I have made some experiments and arrived at a geometry that allows inserting the magnets easily while still keeping them in place a little:
- I added a cubic clearance of around 0.2 around the magnets of the MainCase in all three spatial directions.
- Clearance was added such that one layer worth of filament was left with the original dimensions. That way there is a ridge around the opening that allows to push the magnet in but which holds them in place and keeps them from jumping out. If they are too lose, they can be fixed in place using 2 component acrylic (or other suitable glue).
- Inserting them was a breeze: Have a tower of cubic magnets, push two into the MainCase and two into the USB cover, and they stay there first try.

Attached: Adapted USB cover + matching STL.

[image]
[adapted_plug.zip](https://github.com/openbikesensor/OpenBikeSensor3dPrintableCase/files/6680777/adapted_plug.zip)

gluap, 2021-07-08:
@tobst  what do you think of the above? I have rudimentary openscad scripts to cut this kind of holes into the case, but unless you think it makes sense I'd rather not invest the time to make them "pull-request" ready.

opatut, 2022-01-16:
I guess with the new openscad branch this becomes trivial. I'm closing this ticket in favor of the more general #34.

closed 2022-01-16 by opatut