openbikesensor/OpenBikeSensor3dPrintableCase #17: OBS-Display-A-005_Kabelbinder_Schiene_v0.1.0: Change Side of the Magnet-Pocket

Carolinitzny, 2021-06-06:
The Magnet-Pocket is on the upper side in build direction, where the printer has to bridge. This is prone to inaccuracy and therefore makes putting the second magnet in very hard (if the printing wasn't as clean :)  ).
Would be nice, if the magnet-pocket is on the other side where the bridging is more accessible and easier to clean up.

tobst, 2021-06-06:
This part has been bothering me since i designed it. I would be happy to redesign. Can you provide a simple sketch to clarify your suggestion?

Carolinitzny, 2021-06-06:
[image]
The light green part is for the handlebar, the darker green is the part where the magnets fit in

opatut, 2022-01-16:
Hi @Carolinitzny, your suggestion was applied on the `openscad` branch at least, maybe even before in the legacy files.

[image]

closed 2022-01-16 by opatut