openbikesensor/OpenBikeSensor3dPrintableCase #18: More space for display

gluap, 2021-06-18:
Currently a lot of suppliers of OLED displays seem to be switching to a new OLED display type which is slightly larger than the one previously delivered. Four out of five suppliers we tried delivered the larger display type, with some giving outright wrong measurements for the displays. The displays were electronically compatible with OBS.

The new type display board is 27.5-28mm wide and 28-29mm long, with the glass being as wide as the board, making it impossible to cut the boards down to size.

I believe an increased display opening would already allow for use of both style (small and large) boards in the same display case.

I attached a 3d Model (stl in zip) of the large-type displays as measured. The active area of the oled display is marked as a slight cutout from the display glass that is not present in the actual device.

[largedisplay.zip](https://github.com/openbikesensor/OpenBikeSensor3dPrintableCase/files/6679185/largedisplay.zip)

opatut, 2022-01-16:
Is this just a clearance problem or do we need a different geometry? I don't understand the new shape, maybe you can provide pictures, @gluap ?

gluap, 2022-01-16:
The overall geometry is the same, the difference is only the display board (and glass) width and height. You can find a working case for such displays here: https://github.com/gluap/OpenBikeSensor3dPrintableCase/tree/oversize_display/DisplayCase/SlideOnDisplay/SpecialVersions .

Essentially having the display width and height parametrized (and the hole through which one looks at the display as well) would allow for this. Clearance alone won't do it properly as the scale factors in the two dimensions of the rectangle do not necessarily match.

opatut, 2022-01-17:
So these are enough?

```
DisplayCaseTop_pcb_width = 26.5;
DisplayCaseTop_pcb_height = 27.2;
DisplayCaseTop_window_width = 21.5;
DisplayCaseTop_window_height = 12.5;
```

:)

gluap, 2022-01-19:
Yes that looks sufficient!

gluap, 2022-04-10:
with parametrized display size the issue should be solved.

PR #19 (2022-01-26): Adding display case for oversize displays
This is a version of the SlideOnDisplay case with heatset inserts for the oversize displays that we had received from different sellers. I used fusion360 to strategically cut and enlarge the display case while leaving the features interacting with other parts (button, magnet position, handlebar mount) alone.

fixes #18
closed 2022-04-10 by gluap