openbikesensor/OpenBikeSensor3dPrintableCase #20: Welche Dateien sind die richtigen?

derlucas, 2021-09-13:
Hallo Zusammen,
aus der Doku https://www.openbikesensor.org/docs/hardware/general/case/ geht nicht hervor, welche Dateien man drucken muss.

Ich habe dem Rat entsprechend das Vertical Case gewählt und stehe vor der Frage OBS-MainCase A B C D?

Eine kurze Aufstellung der Files für ein Gehäuse samt Halterung an der Sattelstütze wäre irgendwie ganz hilfreich.

Grüße
Lucas

opatut, 2021-09-14:
[Dieser PR](https://github.com/openbikesensor/openbikesensor.github.io/pull/95) wird das vermutlich lösen.

SubOptimal, 2021-09-14:
Du kannst auch das Release Archiv herunterladen. Die Dateien sind für den OpenBikeSensor v00.03.12.

https://github.com/openbikesensor/OpenBikeSensor3dPrintableCase/releases/tag/v0.1.0

opatut, 2022-01-16:
Noch besser wird das ab sofort von https://github.com/openbikesensor/openbikesensor.github.io/pull/129 abgedeckt. Ich schließe mal dieses Ticket.

closed 2022-01-16 by opatut