openbikesensor/OpenBikeSensor3dPrintableCase #21: Add loop for safety strap on main case

gluap, 2021-09-28:
It happened with several of our cases that the holding bracket broke while the driver was cycling on rough ground. So far in all cases the OBS and the Bike suffered no permanent damage. Apart from the broken bracket of course. I have moved to a modified bracket (attached) to reduce likelihood of this happening.

Without a safety strap a broken braket may always result in the OBS falling into the back wheel causing the driver to fall resulting in injury and a potentially damaged bike. To avoid this risk I imagine one could have a loop to attach a safety strap that to be secured against the seatpost. 

[image]

gluap, 2021-09-28:
[strengthened mount.zip](https://github.com/openbikesensor/OpenBikeSensor3dPrintableCase/files/7246949/strengthened.mount.zip)

gluap, 2021-09-29:
I printed a few of the thicker brakets with the simple strap clamp shown above to test them.

[image]

Ventusfahrer, 2021-11-25:
Hi,

I would also suggest to move the chamber more to the top. 

[image]

Despite the current increase of the outer parts of the brackets the critical part is still unchanged (with standard slicer settings 2 outer shells and one for the inner one of the chamber). The critical part is the red marked part of the structure. Here  we have a high drag. At the right border we have pressure and in the middle nearly no force.

Moving the chamber more (or better as most as possible) to the right will move it more to the neutral line and create a higher wall thickness of the most stressed part. 

On the other hand: I strongly support the exchange the standard OBS mount part with the one proposed in this change (may be with my additional proposal)

PR #22 (2021-12-05): Add mount with strengthened bracket.
This is a strenghtened mount including the thickened bracket together with the shifted pin locking hole suggested by @Ventusfahrer. Fixes #21.

The old bracket was known to break in the now-strengthened locations.

At-mention @tobst because he requested this PR.

Caveat: For full usability, this will also need a modification of the build manual to make users aware of how to use the included safety strap fixture.
closed 2021-12-05 by tobst