openbikesensor/OpenBikeSensor3dPrintableCase #27: OpenSCAD: Auto-generate release bundle (STL files) and render PNGs

opatut, 2022-01-07:
We could configure (for the `openscad` branch) a GitHub Workflow that renders the STLs and a bunch of images as PNGs. Maybe we can even embed the PNGs in the readme then to show which parts are which. The STLs should be downloadable as a zip archive.

gluap, 2022-02-28:
One change I recently added that would be helpful for this workflow is that now the vertices in all stl files are sorted. I believe that this allows to have reproducible stl files - hence only changed geometry yields a changed stl.

gluap, 2022-04-10:
I added a github workflow that bilds the STLs, creates a draft release with a zip of the stls and automatically commits the pngs to the repo (so they are in the usual location). If required we can easily modify it to fix the issue. Adding an auto-commit for the stls would also be possible but with repo size in mind I think we should go the other direction and stop serving stls via git altogether.

closed 2022-04-10 by gluap