openbikesensor/OpenBikeSensor3dPrintableCase #40: Create TPU printable seal for display glass

opatut, 2022-01-17:
The display glass on my module is raised 1.4mm above the PCB surface. The `DisplayCaseBottom_standoff_height` is 2mm. So there is a 0.6mm gap that would be perfect for a little TPU-printed seal :)

Let's design one and try that out. Sure, not everybody can print TPU, but if we can centrall print a few hundred of them, and ship them around, then probably everybody can have one ;)

gluap, 2022-01-17:
I have some TPU around and still have the modified extruder part installed to print it. It takes surprisingly little effort to print TPU on an Ender3 (albeit very slowly and with some stringing).

Only downside: The TPU is bright green.

opatut, 2022-01-18:
Hm, I tried this, and I'm not happy with it. I think my TPU (NinjaTek Cheetah, 95A Shore) is too hard for good seals :(

I'd also still have to work on the geometry. My display should have the window shifted by 1.2-1.5mm, not 2mm, to the bottom. I will need to do some more test printing. Maybe even smaller layer height might be a good idea, then print maybe 0.7mm instead of 0.6mm to get some compression.

Basically, it'll take more time for experiments ;)

opatut, 2022-01-18:
Some day I'll try to 3D print a template or mold, then cast something in 2-part silicone. That would probably make a very nice seal :)