openbikesensor/OpenBikeSensor3dPrintableCase #41: Let makefile know about build dependencies

gluap, 2022-01-17:
The makefile is great and on a multicore machine ``-j`` will speed up the build considerably. But I noticed that make only detects direct 1:1 dependencies between the one source file built and its build target. e.g. the following will **not** render the changes:

```
make
vim variables.scad # change something
make
```
because make dos not know that ```variables.scad``` is depended on by the other files and hence a change in ```variables.scad``` necessitates a full rebuild.

Could be solved by make always buillding everything as I suspect it may be too much of a hassle to properly reflect dependencies.

opatut, 2022-01-18:
> Could be solved by make always buillding everything as I suspect it may be too much of a hassle to properly reflect dependencies.

But then all STLs and PNGs will be overwritten all the time even if nothing changes. That's going to be bad for the git history. I don't think the build output is stable / deteministic :(

gluap, 2022-01-18:
> I don't think the build output is stable / deteministic :(
It's not deterministic, at least not between different versions of openscad. For Full rebuilds I always see geomatrically unchanged files changing. I then nanually submit only those that I actually have changed.

Long-term solution may be: ``.stl`` files live in a separate repo and are rebuilt by a git action build step that pushes them there. That way we'd have one repo with a cluttery history containing only the STLs and another with a clean history of the sources. Alternatively we could leave the STL files entirely out of the repo and upload them to the web page once the web page has that capability.

In general I think that having build outputs in the repo is not necessarily a good idea as one often faces the problem about non-reproducible builds and exceedingly spammy history as a result.

opatut, 2022-01-18:
#27 should solve that. We can use github releases for now I guess. The thumbnails don't need to be updated too often I think.

gluap, 2022-04-10:
We now have a github wofkflow that auto-builds the stls into a release zip - And on top of that we have an stl sorter that minifys the stl diffs to only leave geometry-changed stls as diffs.

closed 2022-04-10 by gluap