openbikesensor/OpenBikeSensor3dPrintableCase #48: New OpenSCAD designs lack bridge helpers

codingcatgirl, 2022-02-22:
I love that you converted everything to OpenSCAD and cleaned up the repo, really awesome!

However, the new versions can't be printed without supports because bridge helpers are missing in crucial places. I've attached two examples where bridging helpers would be needed. This definitely confuses me because in `variables.scad` there's an `enable_easy_print` option which sounds like it should add bridge helpers but it looks like it doesn't, at least not in these places?

I've attached two pictures showing areas with issues and two pictures of the bridge helpers in the old version of the case for reference.

[image]
[image]
[image]
[image]

codingcatgirl, 2022-02-22:
Was just about to submit a pull request, then noticed it's fixed in #47 :) Cool, i'll just use that branch then.

codingcatgirl, 2022-02-22:
It is also fixed for the switch, however at the GPS hole there is still a t-shaped bridge left. Not that big a deal, but probably good to fix as well :)

gluap, 2022-02-22:
@codingcatgirl So the below screenshot is the problematic bridge, right?

To avoid double work: are you looking into it? Otherwise I'd see that I do it later.

If you are in a hurry: For me that corner wasn't beautiful but didn_t look it had structural problems either.

[image]

codingcatgirl, 2022-02-22:
Hey! No, i'm not looking into it right now. This one could be easily fixed by SuperSlicer's bridging features. And even if it wasn't, since it's on the inside it would have bothered me less. :)

closed 2022-02-26 by gluap