openbikesensor/OpenBikeSensor3dPrintableCase #52: Critical: PCB doesn't fit into currently published designs

codingcatgirl, 2022-02-25:
What it says in the title. The current files are unusable. I've already printed two (of fifteen) and luckily just noticed early that the PCB doesn't fit. I also have an old case and it fits in there. Using the DebugPCB() module in the OpenSCAD files i can even confirm that the design ist wrong.

**WTF??**

Have the OpenSCAD files not been tested? 
Should i expect more problems like this? 
Should i just use the legacy files for now? Should everyone?

This is instantly noticable if you try to fit the PCB into the case. I'm glad i only found out about this after two printed cases and not later – and it looks like the PCB can at least be sanded down on the side in question. Still, **this needs urgent fixing before anyone else prints broken parts!**

Some illustrations of the problem using the DebugPCB in OpenSCAD. In the current design **the PCB is intersects with the wall**:

This is how it looks right now on the main branch:
[image]

This is how it looks if i move the PCB 1mm out of that wall:
[image]

gluap, 2022-02-26:
Very rightfully WTF. This has been in the branch for quite some time, so I'll add this warning in the forum etc.. later on. Will push a fix soon. Thank you for the issue and apologies for not noticing - with hardware devices put together and we felt safe to push this. It seems that both @opatut s and my pcb were smaller (mine was an older minor revision, likely explaining the discrepancy.  But @tobst also has a 0.03.12 PCB that shows the issue and the kicad dimensions match the board model in the repo.

closed 2022-02-26 by gluap

gluap, 2022-02-26:
@codingcatgirl One question that comes with the merge: Were your prints from the MAIN or from the scad-generator branch? If they were from MAIN the lid won't fit after the merge - I could backport the fix to the old MAIN branch though if you already have a bunch of lids that you want to use.

codingcatgirl, 2022-02-26:
I have printed stuff from the scad-generator branch and i'm always printing the lid and main case together, so there should be no issue with that part. I found the problem with the models from the scad-generator branch and then confirmed them to be present on the main branch. So… your merge right now did not fix the problem yet!

gluap, 2022-02-26:
@codingcatgirl the issue was that my push didn't go through - I fixed in the scad generator branch but overlooked my git command line telling me that my public key was rejected. After pushing with re-passworded keepass agent it should now be fixed in both branches.

codingcatgirl, 2022-02-26:
Haha, i figured it might have been something like that. Let me quickly check if it looks alright on my side as well!

codingcatgirl, 2022-02-26:
That looks much better! However… is it intentional that the USB-C port isn't centered in its cutout now
[image]
?

codingcatgirl, 2022-02-26:
Also, interesting idea for the future: This might overwhelm openSCAD, but if it doesn't you could try to calculate an intersection of the PCB and the case and see if it results in an empty geometry as a safety check as part of the build/make process.

gluap, 2022-02-26:
@codingcatgirl @tobst tested the change with the latest board and reported that the space near the sensor was a bit tight (though he was able to put it together) - I pushed a fix for that as well. He's test-printing the design overnight. [status is in this forum thread](https://forum.openbikesensor.org/t/achtung-gehaeusedrucker-issue-im-maincase-von-januar-bis-heute/586/11)

codingcatgirl, 2022-02-27:
cool, thanks! i'll probably continue printing cases tomorrow, waiting for some filament to be delivered right now :D