openbikesensor/OpenBikeSensor3dPrintableCase #55: Curved bridging since latest commits

codingcatgirl, 2022-02-28:
One of the bridge helpers at the switch port has a curved perimeter, which is not ideal and defeats the purpose a bit :)

[image]

gluap, 2022-02-28:
I take this to mean "I expected an octagonal final layer of bridge helpers", is that right? 
I stopped at a square like we had in the original design (below).

[image]

codingcatgirl, 2022-02-28:
no, i specifically mean the curved perimeter aliend with the outside shape of the case (top right corner) :) That wasn't like that a few days ago i think.

gluap, 2022-03-01:
I added the chage to the beta branch, it should now not be curved any more. This is what it looks sliced for me in the beta version.
[image]

codingcatgirl, 2022-03-04:
Sounds great. I would recommend merging it soon since i get quite a bit of a mess in this section of the part without the fix.

gluap, 2022-03-04:
It is already available [here](https://github.com/openbikesensor/OpenBikeSensor3dPrintableCase/tree/beta) - I also have a printed version that goes together, but to avoid issues with non-fitting PCBs in the future I'd rather wait until the first one of these has been built successfully (with the PCB 0.03.12, which I don't have yet.).

codingcatgirl, 2022-03-04:
With my slicing settings it looks much worse with the curve. I might have to look into that. Anyway, yes i know it's on the beta branch, it's what i'm currently printing :)

gluap, 2022-04-10:
I merged the fix to main with the latest merge from beta now that there are several cases out there with the fix.

closed 2022-04-10 by gluap