openbikesensor/OpenBikeSensor3dPrintableCase #67: Add more default clearance

opatut, 2022-07-11:
Especially these parts, when not taken enough care of while printing, will not fit well together, so let's add more clearance:

* StandardMountAdapter / any mount (these hardly ever fit)
* USBCover / MainCase (0.2mm more will probably do)
* GPSAntennaLid / MainCase (the notch in the MainCase should be a bit wider, deeper and thicker)

gluap, 2022-10-09:
- [ ] StandardMountAdapter sliced with a current cura version fits perfectly - and I assume the fit is improved for latest prusa as well as they have switched to arachne.  I'd suggest to not change this just now because being too lose is way worse than being too tight and needing a bit of filing.
- [x] USBCover was implemented by you in 487eaf1272868e120a3e776d9efa801df8db85eb :partying_face: 
- [x] GPSAntennaLid/MainCase added default_clearance to AntennaLid in d9bd1c09eff34cf3a96ab3ed9d44ad6af05f88eb

gluap, 2023-01-04:
The change is now merged to main - closing the issue.

closed 2023-01-04 by gluap